MHEtLive/ESP32-MINI-KIT #3: Schematics?

kzyapkov, 2017-11-27:
I am integrating the MiniKit in a device with weird power supply, which would be much simpler if I had the schematics. Is there a protection diode for USB power? What are the 3v3 LDO specs, can it be fed with 12V?

But yeah, just release the schematics. All other devkits do it.

MHEtLive, 2017-11-27:
The LDO 4.75V ≤ VIN ≤ 12V,but notice that 0 ≤ IOUT ≤ 800mA ,

sschueller, 2018-03-17:
I believe this is the schematic
[image]

MHEtLive, 2018-06-26:
Yes.@sschueller   And you can find somthing new-update from here about mh-et live minikit.
http://forum.mhetlive.com/topic/8/mh-et-live-minikit-for-esp32

closed 2018-06-26 by MHEtLive

kakaducsy, 2018-12-07:
@MHEtLive 
I think schematic is good, but not flexible. This Kit only use with USB cable, cannot use with batteries

pierremolinaro, 2019-05-14:
This is NOT the schematic :
  - MH ET LIVE minikit has only one push button, the schematics shows two;
  - IO2 builtin led is active high, the schematic shows an active low led.

MuchammadSobrun, 2019-07-05:
That is the schematic of MH-ET LIVE ESP32 Development Board WiFi + Bluetooth. not minikit.

pierremolinaro, 2019-07-06:
Okay, however the schematic's title is MH-ET LIVE ESP32-MINIKIT…

And where can we find the minikit schematic ?

Pierre

ASTEG012, 2020-01-15:
It is very difficult to find relatable documentation for the MiniKit and that what you find is not always very clear. I am still not sure if the MiniKit development board can be powered with external 5V as the pin layout plan suggest. 

Did any body has experiences with this board?

ricardoquesada, 2020-10-05:
it is possible to power the minikit with 5v (i'm doing that)... but regardless of that, I'm still interested in the minikit schematic... any update @MHEtLive ?

kekersdev, 2021-08-24:
> I believe this is the schematic
> [image]

It looks lite there is an error in the auto program/reset circuit (bottom right corner): DTR is connected to both bases and RTS to both emitters but instead it should be connected like this 
[image]