MHEtLive/ESP32-MINI-KIT #8: Footprint for eagle

beckmx, 2020-07-22:
Hello, do you happen to have a footprint for eagle of this board?

MHEtLive, 2020-07-23:
sorry.No eagle footprint.

sabifa, 2020-11-25:
Any update on this?

zimmer62, 2020-12-16:
I'd like an Eagle footprint also...

thomasdg, 2021-01-07:
Me too, I just ordered a bunch of those board before checking in eagle if a footprint existed and feel sorry now... Please help uw make nice PCBs for use with your boards!

sabifa, 2021-01-11:
I've created a footprint myself as I couldn't find something. You can find the library here: https://grabcad.com/library/mh-et-esp32-d1-mini-eagle-1

Please verify the pins and their position before ordering PCBs!

thomasdg, 2021-01-13:
Thanks a lot for sharing this! 
I was already looking for tutorials on creating my own :)

terragady, 2021-02-24:
Did anyone check if this footprint is correct?

thomasdg, 2021-02-24:
I used it in a PCB I created and the system works perfectly.
Now have the PCBs in production with some more pins used (like, almost all).
When I looked over the pin names in eagle vs the real board, everything kept linking correctly.

Thomas

terragady, 2021-02-24:
perfect thanks