MHEtLive/ESP32-MINI-KIT #9: GPIO pin mismatch

guneshuseyin, 2020-11-25:
Hello,

Im trying to add 2 digital output from GPIO22 and GPIO26 pins like;

#define No1 22 //Also i tried #define No1 GPIO22 // #define No1 D1
#define No2 26

But this lines result is not matching on board pins. 
What i need to write for controlling D0 and D1?

solroshan, 2021-03-05:
> Hello,
> 
> Im trying to add 2 digital output from GPIO22 and GPIO26 pins like;
> 
> #define No1 22 //Also i tried #define No1 GPIO22 // #define No1 D1
> #define No2 26
> 
> But this lines result is not matching on board pins.
> What i need to write for controlling D0 and D1?

It's look like your board configuration's mismatch.

guneshuseyin, 2021-03-06:
> It's look like your board configuration's mismatch.

@solroshan Thank you! Its true, i made my board selection wrong. Its fixed after select right board.

closed 2021-03-06 by guneshuseyin