MHEtLive/ESP32-MINI-KIT #13: Failed to write to target RAM problem

hozdemir, 2021-09-30:
when i upload the sketch into board it gives the following error;

esptool.py v3.0-dev
Serial port /dev/cu.usbmodem52870179151
Connecting........_
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:61:05:12:1c:a4
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000)
A fatal error occurred: Failed to write to target RAM (result was 01070000)

i'm using macos m1 big sur.

how can i solve this?

tilman, 2021-10-29:
I have the same error, also with an ESP32 MiniKIT V2.0 from MH-ET Live. Have you found a way to fix it?
Also I have tried to use an external FTDI Adapter but without success as well.

tilman, 2021-10-29:
I managed to get it running using the following setup:
* an external FTDI Adapter
* IO0 connected to GND
* 2.2 uF Capacitor between RST and GND
* USB Cable plugged into a separate power supply

SALTCO, 2022-08-12:
if you have the error on macos, you can try this:
https://github.com/WCHSoftGroup/ch34xser_macos